EmotiBit/ofxEmotiBit #5: Add functionality to catch LSL triggers and pass them to the EmotiBit

produceconsumerobot, 2019-06-12:


produceconsumerobot, 2019-06-12:
I found an example code online that mimics a simple experiment where the GUI displays "Hello" and "World" alternately, while sending triggers to LSL at each word onset.

EmotiBit, 2019-06-13:
[image] [https://github.com/EmotiBit/ofxEmotiBit/issues/5](https://trello.com/c/lIPR3WsO/32-https-githubcom-emotibit-ofxemotibit-issues-5)

produceconsumerobot, 2019-06-17:
- LSL github https://github.com/sccn/labstreaminglayer
- LSL info https://www.youtube.com/watch?v=Y1at7yrcFW0
- Trigger test code https://github.com/kaczmarj/psychopy-lsl/blob/master/example_coder.py

badfishblues, 2019-06-17:
- cpp library: https://github.com/sccn/liblsl

produceconsumerobot, 2019-06-19:
Use this OF addon as a starting point for adding LSL support to our oscilloscope https://github.com/moostrik/ofxLSL

closed 2019-07-02 by badfishblues