EmotiBit/ofxEmotiBit #34: Parser: Create fallback for case when no/few sync pulses are detected

produceconsumerobot, 2020-02-04:


nitin710, 2020-05-28:
## Solving timeSync issues

### Have an onBoard timing control
1. **Use MAX30101 to generate sync Pulses**
    - The MAX, after it is set to work at a particular rate, can be used to generate INT pulses.
    - Right now, the INT is connected to an External Interrupt pin([SAMD21, Pg.325 section-21.1](http://ww1.microchip.com/downloads/en/DeviceDoc/SAM_D21_DA1_Family_DataSheet_DS40001882F.pdf)) on the feather. We can use that INT pulse to periodically sync and correct the timing on the chip.
2. **Use the RTC clock onBoard**
    - Slow and unreliable according to online shatter,[arduino forumdiscussion](https://forum.arduino.cc/index.php?topic=458391.0)
    - 

### Depend on NTP for time synchronization
#### Requirements of this architecture
- Connection to the Internet
  - Long enough connection to the internet
#### Possibilities
1.  **Always send a timeSync when beginning record/ending Record**
    - **Issue**- Can not guarantee that the emotibit will ever again connect to the internet for the second timeSync

nitin710, 2023-01-19:
solved with https://github.com/EmotiBit/ofxEmotiBit/releases/tag/v1.4.11

PR #279 (2025-08-08): Fix xcode proj files
# Description

- Fixes to xcode project files to conform to new xcode

# Requirements

- None

# Issues Referenced

- None

# Documentation update
- None

# Testing

## Feature Tests (Integration tests)

|Feature Test | Result|
|--------------|-------|
|Feature test 1| ✔️ or ✖️|

## Unit Tests
closed 2023-01-19 by nitin710