EmotiBit/ofxEmotiBit #99: DataParser: Verify all `RD` expected outcomes are being covered

nitin710, 2022-04-14:
# Description
EmotiBit sends an `RD` packet to the host to get timeSyncs.
- A key point to note, is that the `RD` packet is sent twice.
- Once directly through UDP
  - https://github.com/EmotiBit/EmotiBit_FeatherWing/blob/c1b057ad47a67b620d53924bef54eec50ecc1974/EmotiBitWiFi.cpp#L547
- Second time 
  - it is added to the main SD record out stream
  - https://github.com/EmotiBit/EmotiBit_FeatherWing/blob/c1b057ad47a67b620d53924bef54eec50ecc1974/EmotiBitWiFi.cpp#L550

So, for a single RD, it is possible to receive multiple `TL`-`AK` pairs

Example:

34909 | 1818 | 2 | RD | 1 | 100 | TL | TU|
-- | -- | -- | -- | -- | -- | -- | --
34921 | 1819 | 1 | TL | 1 | 100 | 2022-02-03_15-36-33-427125 |  
34930 | 1820 | 2 | AK | 1 | 100 | 918 | RD
34938 | 1821 | 1 | TL | 1 | 100 | 2022-02-03_15-36-38-612039 |  
34942 | 1822 | 2 | AK | 1 | 100 | 1818 | RD|

`34909 | 1818 | 2 | RD | 1 | 100 | TL | TU|`  is the original `RD`

TL response for a previous RD(Note how the RD packet number in AK does not match with the RD
```
34921 | 1819 | 1 | TL | 1 | 100 | 2022-02-03_15-36-33-427125 |  
34930 | 1820 | 2 | AK | 1 | 100 | 918 | RD
```

TL response for the current RD(the RD packet number in AK matches(1818)
```
34938 | 1821 | 1 | TL | 1 | 100 | 2022-02-03_15-36-38-612039 |  
34942 | 1822 | 2 | AK | 1 | 100 | 1818 | RD|
```

We need to make sure we can account for all cases of RD-TL messaging so as to avoid any errors in parsing

# Possible RD-TL scenarios

1. Expected  [✔️ ]timeSyncTableUpdatedCorrectly  | [✔️ ]parsed correctly
```
29801,918,2,RD,1,100,TL,TU
29820,919,1,TL,1,100,2022-02-03_15-36-33-359625
29823,920,2,AK,1,100,918,RD
```
--------------------------------------------------------------------------------------
2. 2 or mode TL received for 1 RD   [✔️ ]timeSyncTableUpdatedCorrectly  | [❌ ]parsed correctly(#97 fixes this)
```
34909,1818,2,RD,1,100,TL,TU
34921,1819,1,TL,1,100,2022-02-03_15-36-33-427125
34930,1820,2,AK,1,100,918,RD
34938,1821,1,TL,1,100,2022-02-03_15-36-38-612039
34942,1822,2,AK,1,100,1818,RD
```
--------------------------------------------------------------------------------------
3. No AK from host   [✔️]timeSyncTableUpdatedCorrectly  | [✔️]parsed correctly
```
29801,918,2,RD,1,100,TL,TU
29820,919,1,TL,1,100,2022-02-03_15-36-33-359625
[UDP loss]
```
RTT will be -1, so this packet will not be considered
--------------------------------------------------------------------------------------
4. No TL from host   [❌]timeSyncTableUpdatedCorrectly  | [❌]parsed correctly #97
```
29801,918,2,RD,1,100,TL,TU
[UDP loss]
29823,920,2,AK,1,100,918,RD
```
https://github.com/EmotiBit/ofxEmotiBit/blob/2ea8b07a6422bf42fcf4e0883b1729fe8752abb0/EmotiBitDataParser/src/ofApp.cpp#L494
this would result in a negative RTT.
We sort the list of RTTs, so having a `-ve` in the list would be flagged as the smallest RTT(wrong)
which would cause erroneous output as this point would be chosen to create the time mapping, but it does not have any TL sent from the host
Fix: **remove any -ve RTT from the list of RTTs**
--------------------------------------------------------------------------------------
5. Multiple responses, with any **but correct** TL missing   [✔️]timeSyncTableUpdatedCorrectly  | [❌ ]parsed correctly #97
```
34909,1818,2,RD,1,100,TL,TU
[UDP loss]
34930,1820,2,AK,1,100,918,RD
34938,1821,1,TL,1,100,2022-02-03_15-36-38-612039
34942,1822,2,AK,1,100,1818,RD
```
--------------------------------------------------------------------------------------
7. Multiple responses, with **the correct TL** missing   [❌]timeSyncTableUpdatedCorrectly  | [❌]parsed correctly #97
```
34909,1818,2,RD,1,100,TL,TU
34921,1819,1,TL,1,100,2022-02-03_15-36-33-427125
34930,1820,2,AK,1,100,918,RD
[UDP loss]
34942,1822,2,AK,1,100,1818,RD
```
The correct AK would acknowledge a wrong TL.
**ToSolve**
--------------------------------------------------------------------------------------
8. Multiple responses, with any **but correct** AK missing   [✔️ ]timeSyncTableUpdatedCorrectly  | [❌]parsed correctly #97
```
34909,1818,2,RD,1,100,TL,TU
34921,1819,1,TL,1,100,2022-02-03_15-36-33-427125
[UDP loss] 34930,1820,2,AK,1,100,918,RD
34938,1821,1,TL,1,100,2022-02-03_15-36-38-612039
34942,1822,2,AK,1,100,1818,RD
```
--------------------------------------------------------------------------------------
9. Multiple responses, with any **the correct** AK missing   [✔️]timeSyncTableUpdatedCorrectly  | [❌ ]parsed correctly #97
```
34909,1818,2,RD,1,100,TL,TU
34921,1819,1,TL,1,100,2022-02-03_15-36-33-427125
34930,1820,2,AK,1,100,918,RD
34938,1821,1,TL,1,100,2022-02-03_15-36-38-612039
[UDP loss]
```
--------------------------------------------------------------------------------------
10. No response
[✔️]timeSyncTableUpdatedCorrectly  | [✔️ ]parsed correctly

nitin710, 2022-04-20:
Tests
|Case number| Case description| is timeSync parsed correctly? | is timeSync file written correctly?|Solution|
|--------------|-------------------|---------------------------------|---------------------------------|-----|
|1|Expected | ✔️  | ✔️  | |
|2| 2 or mode TL received for 1 RD |  ✔️ | ✔️ #97  | - |
|3|No AK from host | ✔️| ✔️ #97 | - |
|4| No TL from host | ✔️ | ✔️  #97 |check on -ve RTT|
|5|Multiple responses, with any but correct TL missing| ✔️ | ✔️ #97  | - |
|6|Multiple responses, with the correct TL missing | ✔️ | ✔️  #97 | Added `else` to check for mis-matched AK|
|7|Multiple responses, with any but correct AK missing | ✔️ | ✔️ #97 | - |
|8|Multiple responses, with the correct AK missing| ✔️ | ✔️ #97 | - 
|9|No response | ✔️ | ✔️ | - |

nitin710, 2022-04-20:
- the file has been modified, where particular TL and AK packets have been deleted to simulate test conditions.
- New rows with typetag DN(Debug Notes) have also been added to indicate the change that has been made
  - [2022-04-20_12-10-28-051739.csv](https://github.com/EmotiBit/ofxEmotiBit/files/8526109/2022-04-20_12-10-28-051739.csv)

- the result of parsing:
  - [image]
  - As seen, we pass all cases, except `case 10`, which i think can only be solved by an Oscilloscope change

produceconsumerobot, 2022-04-28:
@nitin710 table and cases is a really great idea and potentially super helpful, though some of the wording is unclear and there are some typos that make parsing the content hard.
"2 or mode TL received" typo
"with any but correct TL missing" is very hard to understand. "lost packets with correct TL present" would be much clearer. Similar for AK case.

PR #106 (2022-04-29): Fix data parser
# Description
Fixes for data parser

# Issues referenced
- Fixes #97 
- Fixes #99
- Fixes #100 
- Fixes #101 
- Fixes #103 
- Fixes #104 
- Fixes #105 
- After Review 1
- Fixes #110 
- Fixes #111 

# Tests
- Refer Each issue to review the test and corresponding result
commit b33bc717 (2022-04-29): fixed #99
closed 2022-04-29 by nitin710