EmotiBit/ofxEmotiBit #111: DataParser: Improve user instructions when too few timesyncs are found

nitin710, 2022-04-28:
# Description
If too few time syncs are found, add user instructions to better educate people how to improve data collection

nitin710, 2022-04-28:
Output when you get less than 2 timestamps
[image]

produceconsumerobot, 2022-04-28:
WARNING: Data file was parsed with less than 2 time-sync events, which can reduce the timestamp accuracy.

EmotiBit periodically generates time-sync events while a connection is established with the EmotiBit Oscilloscope software. At a minimum, it's recommended to keep EmotiBit connected to the EmotiBit Oscilloscope software for at least one minute after starting data recording AND re-establish connection with EmotiBit Oscilloscope software (using the same computer on which recording was started) for at least one minute before stopping data recording. 

To further improve timestamp accuracy, it's optimal to keep EmotiBit connected to the EmotiBit Oscilloscope software throughout recording to generate many time-sync events in the data file.

nitin710, 2022-04-29:
[image]

PR #106 (2022-04-29): Fix data parser
# Description
Fixes for data parser

# Issues referenced
- Fixes #97 
- Fixes #99
- Fixes #100 
- Fixes #101 
- Fixes #103 
- Fixes #104 
- Fixes #105 
- After Review 1
- Fixes #110 
- Fixes #111 

# Tests
- Refer Each issue to review the test and corresponding result
closed 2022-04-29 by nitin710