EmotiBit/ofxEmotiBit #112: DataParser: Improve algorithm used to choose RTT to create time sync map

nitin710, 2022-04-29:
# Description
- maximize the distance between timestamps
- minimize RTTs

produceconsumerobot, 2022-05-10:
// Cases Best to Worst
	// Q1 : Q4
	// Q1 : Q3
	// Q2 : Q4
	// Q2 : Q3
	// Q1 : Q2
	// Q3 : Q4
	// Q2 : Q2
	// Q3 : Q3
	// Q1 : Q1
	// Q4 : Q4

produceconsumerobot, 2022-05-10:
# Testing
- [ ] Test all cases with a manually modified 4 minute file with AKs deleted from specific quartiles
- [ ] Look at timestamp histograms of a "full" and a Qa/Qb and a Qa/Qa file in python

nitin710, 2022-05-11:
Test Cases Generated: 
Cases Best to Worst
// Q1 : Q4 ✔️ 
// Q1 : Q3 ✔️ 
// Q2 : Q4 ✔️ 
// Q2 : Q3 ✔️ 
// Q1 : Q2 ✔️ 
// Q3 : Q4 ✔️ 
// Q2 : Q2 ✔️ 
// Q3 : Q3 ✔️ 
// Q1 : Q1 ✔️ 
// Q4 : Q4 ✔️

nitin710, 2022-05-12:
# Testing

|Test Case | Quadrant selected | Test Result | Notes|
|-----------|-------------------|-------------|---------|
| Q1 : Q1 | Q1:Q4 | ❔| |
| Q1 : Q2 | Q1:Q4 | ❔| |
| Q1 : Q3 | Q1:Q4 | ❔ | |
| Q1 : Q4 | Q1:Q4 | ❔ | |
| Q2 : Q2 | Q1:Q4 | ❌ |Only 1 RTT(5000mS) was stored in Q1 which was then selected for generating the map |
| Q2 : Q3 | Q1:Q4 | ❔ | |
| Q2 : Q4 | Q1:Q4 | ❔| |
| Q3 : Q3 | Q1:Q4 | ❌ | Only 1 RTT(5000mS) was stored in Q1 which was then selected for generating the map |
| Q3 : Q4 | Q1:Q4 | ❔| |
| Q4 : Q4 | Q1:Q4 | ❌ | Only 1 RTT(5000mS) was stored in Q1 which was then selected for generating the map |

PR #122 (2022-05-13): Improved timesync map algorithm. Removed cmd line timesync warning hang.
# New Features
- closes #112
- closes #115
commit 45fc306d (2022-05-13): Quartile sync map seemingly working. Closes #112.
closed 2022-05-13 by nitin710