EmotiBit/ofxEmotiBit #125: DataParser overwrites data file if no timestamps are available

unidesigner, 2022-05-18:
**Describe the bug**
DataParser overwrites data file if no timestamps are available. The problem was that I did not wait long enough before and after the session for a timesync event to occur (how to make sure that they actually occured?). 

**To Reproduce**
Steps to reproduce the behavior:
1. So when opening the csv file in the current DataParser v1.4.11

**Expected behavior**
Just see an error message, not overwrite the csv file. Also load the data file with "inaccurate timestamps perhaps?

**Screenshots**
```
bin$ ./EmotiBitDataParser 
Removing file: Data/2022-05-18_12-24-08-143587.csv
Creating file: Data/2022-05-18_12-24-08-143587_timesyncs.csv
Processing file: 2022-05-18_12-24-08-143587_timesyncs.csv
Creating file: Data/2022-05-18_12-24-08-143587_timesyncs_timesyncs.csv
**** MALFORMED PACKET 1: 0, 6: RD,TS_received,TS_sent,AK,RoundTrip, ****
**** MALFORMED PACKET DATA: RD,TS_received,TS_sent,AK,RoundTrip,
**** MALFORMED PACKET 2: 0, 6: RD,TS_received,TS_sent,AK,RoundTrip, ****
**** MALFORMED PACKET DATA: RD,TS_received,TS_sent,AK,RoundTrip,
**** MALFORMED PACKET 3: 0, 6: RD,TS_received,TS_sent,AK,RoundTrip, ****
**** MALFORMED PACKET DATA: RD,TS_received,TS_sent,AK,RoundTrip,
**** MALFORMED PACKET 4: 0, 6: RD,TS_received,TS_sent,AK,RoundTrip, ****
**** MALFORMED PACKET DATA: RD,TS_received,TS_sent,AK,RoundTrip,
**** MALFORMED PACKET 5: 0, 6: RD,TS_received,TS_sent,AK,RoundTrip, ****
**** MALFORMED PACKET DATA: RD,TS_received,TS_sent,AK,RoundTrip,
**** MALFORMED PACKET 6: 0, 6: RD,TS_received,TS_sent,AK,RoundTrip, ****
**** MALFORMED PACKET DATA: RD,TS_received,TS_sent,AK,RoundTrip,
**** MALFORMED PACKET 7: 0, 6: RD,TS_received,TS_sent,AK,RoundTrip, ****
**** MALFORMED PACKET DATA: RD,TS_received,TS_sent,AK,RoundTrip,
**** MALFORMED PACKET 8: 0, 6: RD,TS_received,TS_sent,AK,RoundTrip, ****
**** MALFORMED PACKET DATA: RD,TS_received,TS_sent,AK,RoundTrip,
**** MALFORMED PACKET 9: 0, 6: RD,TS_received,TS_sent,AK,RoundTrip, ****
**** MALFORMED PACKET DATA: RD,TS_received,TS_sent,AK,RoundTrip,
**** MALFORMED PACKET 10: 0, 6: RD,TS_received,TS_sent,AK,RoundTrip, ****
**** MALFORMED PACKET DATA: RD,TS_received,TS_sent,AK,RoundTrip,
[notice ] End of file: 11
[notice ] Total RTT's found: 0
[notice ] calculateTimeSyncMap: 0 timesyncs found

Creating file: Data/2022-05-18_12-24-08-143587_timesyncs_timeSyncMap.csv
[notice ] End of file: 12
[notice ] 0 Timesyncs Found
[notice ] WARNING: Data file was parsed with less than 2 time-sync events, which can reduce the timestamp accuracy.

EmotiBit periodically generates time-sync events while a connection is established with the EmotiBit Oscilloscope software.
At a minimum, it's recommended to keep EmotiBit connected to the EmotiBit Oscilloscope software
for at least one minute after starting data recording AND re-establish connection with EmotiBit Oscilloscope
software (using the same computer on which recording was started) for at least one minute before stopping data recording.

To further improve timestamp accuracy, it's optimal to keep EmotiBit connected to the EmotiBit Oscilloscope software
throughout recording to generate many time-sync events in the data file.

exit()
```

nitin710, 2022-05-20:
Hi @unidesigner ,

I am unable to re-create the bug you have described in this issue. I tested it with a file with 0 timestamps, and it performs as expected. It does not overwrite or modify the original data file.

Also, the output in the console looks weird. Is it possible you tried to parse an already parsed file?
`Data/2022-05-18_12-24-08-143587_timesyncs_timesyncs.csv` suggests you tried to parse `2022-05-18_12-24-08-143587_timesyncs.csv`. Can you confirm?

Also, can you share the file that caused this behavior?

unidesigner, 2022-05-21:
Hi @nitin710,

then, I must have done something wrong, perhaps tried to open the wrong file as you suggest. I will report if it would happen again with a detailed description on the steps.

Thanks, and sorry for the noise.

closed 2022-05-21 by unidesigner