EmotiBit/ofxEmotiBit #150: Add LSL output stream

nitin710, 2022-07-12:
# Requirements
- #148 
- #147

jnvandermeer, 2023-03-06:
Dear developers, I have just compiled the Oscilloscope on Arch (after some debugging and googling). I am looking to have an LSL output stream that I can then use to capture the signals from the Emotibit and store them myself in a  (dynamic) data array. Preferably a stand-along software that does not involve the Oscilloscope. Is any development underway, and is there a way to get started for me to (maybe) contribute? Many thanks
Johan

nitin710, 2023-03-07:
Hi @jnvandermeer ,
Thanks for your interest in EmotiBit!

Adding an LSL output stream requires some refactor in the EmotiBit core on both the software and firmware fronts.

On the software side, it needs unlocking at minimum,  the issues mentioned above.
On the firmware side, it requires enabling TCP data transmission, which requires some refactoring. Additionally, it need to assess the impact on resources on EmotiBit (since TCP can be demanding and EmotiBit is running in an embedded environment).
On a separate note, we are also looking into expanding compatibility with [brainflow](https://brainflow.readthedocs.io/en/stable/index.html#), which involves further changes in the firmware!

We are moving towards that and are working on creating a roadmap. I will be sure to ping on this thread once we have a plan in place. Thanks for offering to contribute!

Also, i do suggest you join our [community](https://www.reddit.com/r/EmotiBit/) to keep updated on EmotiBit. Once we have a public facing plan in place, we will definitely announce on the [forum](https://www.reddit.com/r/EmotiBit/)!

produceconsumerobot, 2024-02-01:
This issue was separated from all of the above listed requirements. Instead the effort was focussed into the following activities detailed in (feat-LslOutput gdoc):

- Adding lslOutputSettings.json
- Adding EmotiBitLsL
- Refactors usage of Patchboard

PR #226 (2024-05-31): Feat lsl output
# Description
- Adds LSL output
- Adds EmotiBitLsl module to handle nuts & bolts
- Refactors usage of Patchboard

# Notes to include with `Release`
The LSL output relays data for each EmotiBit stream plotted in the `EmotiBit Oscilloscope`. Please note, however, that the LSL timestamps are presently not corrected for the device-to-computer time-of-flight because this correction can require some degree of nuance for gracefully handling changes in the clock-to-clock estimation. Recording LSL Marker inputs with `EmotiBit Oscilloscope` followed by using the `EmotiBit DataParser` to  [... cut]
closed 2024-05-31 by nitin710