EmotiBit/ofxEmotiBit #165: Red light on

kurnevsky, 2022-08-08:
Hi. Reddit thinks my messages are spam, and I didn't get any response to my email :(
So here is the only place left where I can ask, sorry that it's not directly related to this repository.

I used to have a working feather with EmotiBit (successfully got data in EmotiBitOscilloscope) but now when I turn on the feather with the connected EmotiBit board the red led on the feather board stays on. No possible reasons are mentioned on https://github.com/EmotiBit/EmotiBit_Docs/blob/master/Getting_Started.md#emotibit-bootup

It happened after I connected the turned on feather with EmotiBit to USB (I still had data in EmotiBitOscilloscope) and pressed reset button (after data stopped coming after some time). I tried to reupload firmware to feather and checked the SD card - everything seems fine except that it doesn't work anymore. What could be the reason?

Here are the logs I get in arduino serial monitor after reset button is pressed:

```
I2C data pin:11
I2C clk pin: 13
hibernate pin: 6
chip sel pin: 19
Firmware version: 1.4.1
vregEnablePinLogic: Active HIGH(V3+)
EmotiBit ready
Setting up I2C For M0...Setting clock to100000Initializing NVM controller: success
```

nitin710, 2022-08-08:
The red led on the Feather is supposed to stay ON. that is normal operation. (as indicated on all the images in the [Bootup section of the documentation](https://github.com/EmotiBit/EmotiBit_Docs/blob/master/Getting_Started.md#emotibit-bootup))

Some qustions:
1. Is that the complete serial log?
2. Are any other lights on the EmotiBit ON?
3. Can you verify if it is connecting/ connected to WiFi?

kurnevsky, 2022-08-08:
1. Yes, it's complete log.
2. No, only red light on Feather. It used to work before, I saw lights on EmotiBit, but now they stay turned off.
3. It's not connecting to the WIFI.

nitin710, 2022-08-09:
It looks like it is not even completing the sensor setup,  (hence you can only see the red feather light).

I'll run it by the team, but if flashing the new FW did not help, It might be a hardware issue. 

I'll post on this thread if i can come up with steps to debug