EmotiBit/ofxEmotiBit #213: Firmware Installer does not work with Feather M0 (Mac M1 Ventura 13.1)

JaeminBBQ, 2023-06-12:
**Describe the bug**
The Firmware Installer is not detecting the EmotiBit with Feather M0.

**Expected behavior**
Detect EmotiBit and install the firmware onto the EmotiBit.

**Additional details**
- EmotiBit software version: 1.8.1
- EmotiBit firmware version: 1.8.1-TA
- EmotiBit hardware version: V04a
- OS version:

**To Reproduce**
Steps to reproduce the behavior:
1. Open Firmware Installer
2. Press Space
3. Press reset on EmotiBit (While not in hib)
4. Connect EmotiBit to computer with data capable cable

**Screenshots**
If applicable, add screenshots to help explain your problem.

| Open Firmware Installer | Press Space | Connect EmotiBit |
| ------------------------ | ------------ | ----------------- |
| [image] | [image] | [image] |

nitin710, 2023-06-12:
Huh, looks the the driver in Apple silicon Macs is either 
1. Not exposing VID/PID in device info
2. VID/PID are not getting registered by the drivers.

Anyway, looks like this can be solved by a software patch. Ill add the fix in #212

nitin710, 2023-06-16:
The Feather M0 shows up as `usbmodem`
and the Feather esp32 shows up as `usbserial` (Apart from the `SLAB_USBtoUART` port)

The solution could require checking for the keyword `usmodem` in the serial name

nitin710, 2023-06-19:
Looks like something in the ofxSerial library may not be handling the information as presented by the drivers in M1 macs: https://github.com/bakercp/ofxSerial/blob/7a02ff97c02fdd038e367806128415ed794e48c7/libs/serial/src/impl/list_ports/list_ports_osx.cc#L241-L277

JaeminBBQ, 2023-06-19:
`When you get the chance, can you share what the PID/VID are for the Adafruit Feather M0, as shown in this link:`
https://www.keychron.com/pages/how-to-check-pid-and-vid-in-your-mac

| Screenshot | Description |
| ----------- | ------------ |
| [image] | Screenshot of Adafruit Feather M0 in USB |

JaeminBBQ, 2023-06-30:
Compiled from source on OpenFramework 0.11.2 but still getting same results where PID and VID are not able to be read.

[image]

nitin710, 2023-07-12:
- Looks like this has already been flagged as an issue here: https://github.com/bakercp/ofxSerial/issues/24
- ofxSerial uses Serial, whjich has this interesting issue: https://github.com/wjwwood/serial/issues/258
- This might offer a solution: https://github.com/wjwwood/serial/pull/271

PR #212 (2023-07-16): Fix firmware installer apple silicon fix
# Description

- Fixes Issue with Firmware Installer on macBooks with apple silicon
- Issues description
  - 2 ports are detected sequentially for a plug-in (one port is through apple drivers, another through SiLabs drivers)
  - The ports do not have the required description, which was used to filter possible Feather from other device plug-ins.
  - Both new ports are sequentially discarded because of description mismatch
The installer keeps waiting for a new plug-in

# Requirements
- ❗ ❗ [update the dependency before compiling] https://github.com/EmotiB [... cut]
closed 2023-07-16 by nitin710