EmotiBit/ofxEmotiBit #219: DataParser parsing error leading to extra columns in parsed file

produceconsumerobot, 2023-08-09:
**Describe the bug**
[image]
From dataset 'G:/.shortcut-targets-by-id/1KogPeL5zzT7nFPtEZ5wjIY4poPyVxgWN/EmotiBit Test Data/XenboX/XenboX at TRI 2023-04-16/Data/Bob/2023-04-16_15-42-28-096912'

produceconsumerobot, 2023-08-09:
This may be a situation where the parser falls behind on writing data

produceconsumerobot, 2023-08-09:
Re-parsed the file and got a similar issue, but in a different location in the file:

> 413257.319560,1681696113.571783,7686739.000,8642,1,EA,1,100,0.030297
413257.386060,1681696113.638283,7686805.500,8659,2,EA,1,100,0.030297
413257.452560,1681696113.704783,7686872.000,8659,2,EA,1,100,0.030297
413257.519560,1681696113.771783,7686939.000,8676,1,EA,1,100,0.030297
413257.586060,1681696113.413258.719560,1681696114.971788,7688139.000,8871,1,EA,1,100,0.030297
413258.786060,1681696115.038288,7688205.500,8886,2,EA,1,100,0.030297
413258.852560,1681696115.104788,7688272.000,8886,2,EA,1,100,0.030297
413258.919560,1681696115.171788,7688339.000,8903,1,EA,1,100,0.030297
413258.986060,1681696115.238289,7688405.500,8919,2,EA,1,100,0.030297

nitin710, 2025-11-21:
outdated.