EmotiBit/ofxEmotiBit #223: Compilation issues with the 11

tobiasweede, 2023-12-16:
Hi,

this is not a bug.
However, others might also be interested.
I try to compile the Oscilator for some time now.
It is definitely not straight forward.

I checkout out the branch fix-OFv0.11.2_2 and use the corresponding OF version.

[image]

Can someone give me some tips how I get the Oscillator running?

nitin710, 2023-12-19:
Looks like an error with openframeworks dependency. 
You mentioned you are using the corresponding OF, i would double check that. The branch is compiling on Windows and I not seen this error before.

Can you share you the folder structure of your project(From OF_ROOT down to the project)? It might be that the EmotiBit Oscilloscope (and ofxEmotiBit) might not be in the right location and missing path to lib files from OF_ROOT.

tobiasweede, 2023-12-28:
Thank you for the suggestions.
I re-ran the ofx library setup script and it seems to work now.
The library download might have been interrupted the first time without me noticing.

nitin710, 2024-01-05:
Closing this issue.

closed 2024-01-05 by nitin710