EmotiBit/ofxEmotiBit #225: Errors Building in PopOS (Ubuntu) 22.04

sebtheiler, 2024-01-03:
**Describe the bug**
I've followed the instructions on the README and have been successful up until I try to run `make` in `(OF_ROOT)/addons/ofxEmotiBit/EmotiBitOscilloscope`. I was able to compile OF, run the polygon example, and the `projectGenerator` CLI is working. Running `make` results in the following errors:

```
Compiling OF library for Release
make[1]: Entering directory '/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/openFrameworksCompiled/project'
Done!
make[1]: Leaving directory '/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/openFrameworksCompiled/project'

Compiling EmotiBitOscilloscope for Release
make[1]: Entering directory '/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxEmotiBit/EmotiBitOscilloscope'
Compiling /home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxEmotiBit/EmotiBitOscilloscope/src/ofApp.cpp
g++ -c -O3 -DNDEBUG -Wall -Werror=return-type -DGCC_HAS_REGEX -std=c++20 -DOF_USING_GTK -DOF_USING_GTK -D_x64 -D_LINUX -Wall -Werror=return-type -DGCC_HAS_REGEX -std=c++20 -pthread -D__LINUX_ALSA__ -D__LINUX_PULSE__ -D__UNIX_JACK__ -D_REENTRANT -pthread -I/usr/include/gstreamer-1.0 -I/usr/include/orc-0.4 -I/usr/include/gstreamer-1.0 -I/usr/include/x86_64-linux-gnu -I/usr/include/opus -I/usr/include/AL -I/usr/include/x86_64-linux-gnu -I/usr/include/rtaudio -I/usr/include/gtk-3.0 -I/usr/include/at-spi2-atk/2.0 -I/usr/include/at-spi-2.0 -I/usr/include/dbus-1.0 -I/usr/lib/x86_64-linux-gnu/dbus-1.0/include -I/usr/include/gtk-3.0 -I/usr/include/gio-unix-2.0 -I/usr/include/cairo -I/usr/include/pango-1.0 -I/usr/include/harfbuzz -I/usr/include/pango-1.0 -I/usr/include/fribidi -I/usr/include/harfbuzz -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/pixman-1 -I/usr/include/uuid -I/usr/include/freetype2 -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/libpng16 -I/usr/include/x86_64-linux-gnu -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/fmod/include -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/glm/include -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/glm/include/glm -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/glm/include/glm/gtc -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/glm/include/glm/simd -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/glm/include/glm/ext -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/glm/include/glm/detail -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/glm/include/glm/gtx -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/json/include -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/kiss/include -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/pugixml/include -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/tess2/include -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/utf8/include -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/utf8/include/utf8 -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/openFrameworks -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/openFrameworks/utils -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/openFrameworks/events -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/openFrameworks/gl -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/openFrameworks/3d -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/openFrameworks/graphics -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/openFrameworks/app -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/openFrameworks/communication -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/openFrameworks/video -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/openFrameworks/sound -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/openFrameworks/types -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/openFrameworks/math -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxEmotiBit/EmotiBitOscilloscope/src -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxGui/src -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxNetwork/src -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxOsc/src -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxOsc/libs -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxOsc/libs/oscpack -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxOsc/libs/oscpack/src -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxOsc/libs/oscpack/src/ip -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxOsc/libs/oscpack/src/ip/posix -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxOsc/libs/oscpack/src/osc -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxXmlSettings/src -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxXmlSettings/libs -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxBiquadFilter/src -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxEmotiBit/src -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxEmotiBit/src/Signal -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxJSON/src -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxJSON/libs -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxJSON/libs/jsoncpp -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxJSON/libs/jsoncpp/include -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxJSON/libs/jsoncpp/include/json -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxJSON/libs/jsoncpp/src -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxLSL/libs/labstreaminglayer/include -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxLSL/src -Isrc -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxNetworkUtils/src -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxNetworkUtils/libs -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxNetworkUtils/libs/ofxNetworkUtils -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxNetworkUtils/libs/ofxNetworkUtils/include -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxNetworkUtils/libs/ofxNetworkUtils/include/ofx -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxNetworkUtils/libs/ofxNetworkUtils/include/ofx/Net -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxNetworkUtils/libs/ofxNetworkUtils/src -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxOscilloscope/src -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxPoco/libs/poco/include -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxPoco/src -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxThreadedLogger/src -I/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/EmotiBit_XPlat_Utils/src -MMD -MP -MF obj/linux64/Release/src/ofApp.d -MT obj/linux64/Release/src/ofApp.o -o obj/linux64/Release/src/ofApp.o -c /home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxEmotiBit/EmotiBitOscilloscope/src/ofApp.cpp
In file included from /home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/glm/include/glm/detail/type_half.hpp:16,
                 from /home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/glm/include/glm/detail/func_packing.inl:5,
                 from /home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/glm/include/glm/packing.hpp:173,
                 from /home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/libs/glm/include/glm/glm.hpp:132,
[... cut ...]
/home/sebtheiler/Code/emotibit/of_v0.12.0_linux64gcc6_release/addons/ofxEmotiBit/EmotiBitOscilloscope/src/ofApp.cpp:591:50: warning: catching polymorphic type ‘class std::exception’ by value [-Wcatch-value=]
[... 265 lines cut ...]
ofxBiquadFilter
ofxEmotiBit
ofxEmscripten
ofxGPS
ofxGui
ofxIO
ofxiOS
ofxJSON
ofxKinect
ofxLSL
ofxNetwork
ofxNetworkUtils
ofxOpenCv
ofxOsc
ofxOscilloscope
ofxPoco
ofxSerial
ofxSvg
ofxThreadedImageLoader
ofxThreadedLogger
ofxUnitTests
ofxVectorGraphics
ofxXmlSettings
```
I've confirmed that `ofxNetworkUtils`, `ofxSerial`, and `ofxIO` are in the `stable` branch.

nitin710, 2024-01-05:
> OF Version: of_v0.12.0_linux64gcc6_release

We have not yet switched to OF0.12.0. The software suite was upgraded to OF 0.11.0. You can grab that from [this link](https://openframeworks.cc/versions/v0.11.0/)

Can you compile with OF 0.11.0 (as mentioned in the [readme](https://github.com/EmotiBit/ofxEmotiBit#requirements)) and let us know if it works?

sebtheiler, 2024-01-05:
Running `./compilePG.sh` with OF 0.11.0 from that link results in the following errors:

```
make: Entering directory '/home/sebtheiler/Code/emotibit/of_v0.11.0_linux64gcc6_release/apps/projectGenerator/commandLine'
Compiling OF library for Release
make[1]: Entering directory '/home/sebtheiler/Code/emotibit/of_v0.11.0_linux64gcc6_release/libs/openFrameworksCompiled/project'
Done!
make[1]: Leaving directory '/home/sebtheiler/Code/emotibit/of_v0.11.0_linux64gcc6_release/libs/openFrameworksCompiled/project'

Compiling projectGenerator for Release
make[1]: Entering directory '/home/sebtheiler/Code/emotibit/of_v0.11.0_linux64gcc6_release/apps/projectGenerator/commandLine'
/home/sebtheiler/Code/emotibit/of_v0.11.0_linux64gcc6_release/libs/openFrameworksCompiled/project/makefileCommon/config.addons.mk:210: *** missing separator.  Stop.
make[1]: Leaving directory '/home/sebtheiler/Code/emotibit/of_v0.11.0_linux64gcc6_release/apps/projectGenerator/commandLine'
make: *** [/home/sebtheiler/Code/emotibit/of_v0.11.0_linux64gcc6_release/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:129: Release] Error 2
make: Leaving directory '/home/sebtheiler/Code/emotibit/of_v0.11.0_linux64gcc6_release/apps/projectGenerator/commandLine'
There has been a problem compiling the command line projectGenerator.
Please report this problem in the forums.
```

Which is why I had tried 0.12.0. Do you know if there is a simple fix for this, or should I report it on the OF forums?

nitin710, 2024-01-08:
Weird, haven't seen that before.
Is there a conflict in the `make` version they want us to use vs what you are using? I don't think they specify anything, so might be a tad tricky to find that out.
Found a [similar issue on github with that suggestion](https://github.com/openframeworks/projectGenerator/issues/237#issuecomment-603716067).

Also, you don't **need** to have a compiled `project generator` to use EmotiBit Software. Once you have downloaded the OF bundle and run scripts to compile the OF libraries, you can just get the required addons and proceed to compile EmotiBitOscilloscope as [documented in the readme](https://github.com/EmotiBit/ofxEmotiBit#ofxemotibit).

Interested to see if that works.

sebtheiler, 2024-01-10:
> Also, you don't need to have a compiled project generator to use EmotiBit Software. Once you have downloaded the OF bundle and run scripts to compile the OF libraries, you can just get the required [...]

I've tried that (and would prefer it to downgrading my make version, though that might potentially work for compiling project generator as I do have make 4.3, which the user in the issue you mentioned also had before downgrading), but I'm getting the following errors:

```
sebtheiler@shuttlecraft:~/Code/emotibit/of_v0.11.0_linux64gcc6_release/addons/ofxEmotiBit/EmotiBitOscilloscope$ make
Compiling OF library for Release
make[1]: Entering directory '/home/sebtheiler/Code/emotibit/of_v0.11.0_linux64gcc6_release/libs/openFrameworksCompiled/project'
Done!
make[1]: Leaving directory '/home/sebtheiler/Code/emotibit/of_v0.11.0_linux64gcc6_release/libs/openFrameworksCompiled/project'

Compiling EmotiBitOscilloscope for Release
make[1]: Entering directory '/home/sebtheiler/Code/emotibit/of_v0.11.0_linux64gcc6_release/addons/ofxEmotiBit/EmotiBitOscilloscope'
/home/sebtheiler/Code/emotibit/of_v0.11.0_linux64gcc6_release/libs/openFrameworksCompiled/project/makefileCommon/config.addons.mk:210: *** missing separator.  Stop.
make[1]: Leaving directory '/home/sebtheiler/Code/emotibit/of_v0.11.0_linux64gcc6_release/addons/ofxEmotiBit/EmotiBitOscilloscope'
make: *** [/home/sebtheiler/Code/emotibit/of_v0.11.0_linux64gcc6_release/libs/openFrameworksCompiled/project/makefileCommon/compile.project.mk:129: Release] Error 2
```

This is after running `./compileOF.sh` and following the rest of the instructions in the README, other than `./compilePG.sh`.

I had thought that compiling the project generator would fix this, but I'm not sure anymore. I can still attempt to downgrade my make version and compile it if you believe that's worth persuing.