EmotiBit/ofxEmotiBit #227: Add thread sleep delay timing control to emotibitCommSettings

produceconsumerobot, 2024-02-01:


produceconsumerobot, 2024-02-13:
# Tests to run

## Independent variables
- 1 EmotiBit, 1 oscilloscope, 1 computer
  - Matrix ThreadSleep CPU
  - 0, 0 RTT
  - Target TS RTT
- 3 EmotiBits, single oscilloscope
  - Target TS CPU
- 3 EmotiBits, 3 oscilloscopes, 1 computer
  - Target TS CPU
- 3 EmotiBits, 1 oscilloscopes per 3 computers
  - Target TS CPU
- 10 EmotiBits, 1 oscilloscope, 1 computer
  - Target TS CPU
- 10 EmotiBits, 1 oscilloscope per 2 computers
  - Target TS CPU

## Dependent variables
- Perfmon CPU  https://techcommunity.microsoft.com/t5/windows-powershell/average-cpu-usage-over-a-few-seconds/m-p/4027458
- RTT

produceconsumerobot, 2024-02-14:
ToDo:
- Code tidyup
- PR

nitin710, 2024-07-11:
Added in v1.11.1

closed 2024-07-11 by nitin710