EmotiBit/ofxEmotiBit #241: Does the parser need the following addons?

nitin710, 2024-06-06:
**Describe the bug**
Why does the parser have the following dependencies:
- biquadFilter
- network
- networkUtils

**Expected behavior**
A clear and concise description of what you expected to happen.

**Additional details**
- EmotiBit software version: 
- EmotiBit firmware version: 
- EmotiBit hardware version: 
- OS version:

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.

nitin710, 2024-06-06:
[image]

Looks like WifiHost in ofxEmotiBit/src need network stuff. We should probably change this organization. Data parser has nothing to do with network stuff. If we can restructure the files, it will reduce the number of deps. parser has in addons

connectedfuturelabs, 2025-03-19:
Fixed with #264

closed 2025-03-19 by connectedfuturelabs